udacity/robotics-beta-test-issues #528: crl Perception, L03E08

zenetio, 2017-07-05:
In RANSAC.py line 5 I have cloud variable but in line 8 I have cloud_in which raises an error. Need change to the same variable name.

https://classroom.udacity.com/nanodegrees/nd209-beta/parts/571766f0-9631-484d-a69f-94ea58110725/modules/e5bfcfbd-3f7d-43fe-8248-0c65d910345a/lessons/8d51e0bf-0fa1-49a7-bd45-e062c4a2121f/concepts/02cbb56e-9e54-4c08-977b-df149cb0bca4

kylesf, 2017-07-05:
I am not sure where you are having a problem. Can you be more specific?

zenetio, 2017-07-05:
In ransac.py I have the following:
`cloud = pcl.load_XYZRGB('tabletop.pcd')

# Voxel Grid filter
vox = cloud_in.make_voxel_grid_filter()`

Note that I am creating a variable named cloud but Voxel is using a variable named cloud_in. So, variable cloud_in does not exists.

sahiljuneja, 2017-07-05:
I think that's already been updated. Can you refresh your page and check again? Both those lessons should have `cloud` as the variable name.

closed 2017-07-06 by sahiljuneja