udacity/robotics-beta-test-issues #533: crl Calibration, L03E15

zenetio, 2017-07-05:
I could not find a valid link to download the file table_scene_lms400.pcd. So, I had to go to PCL site and download the file from there.

https://classroom.udacity.com/nanodegrees/nd209-beta/parts/571766f0-9631-484d-a69f-94ea58110725/modules/e5bfcfbd-3f7d-43fe-8248-0c65d910345a/lessons/8d51e0bf-0fa1-49a7-bd45-e062c4a2121f/concepts/fdb3a445-43e0-4a02-81e2-0448432c156f

sahiljuneja, 2017-07-05:
The tabletop.pcd is provided in the github repo for the exercise. 

Here's the direct link - https://github.com/udacity/RoboND-Perception-Exercises/tree/master/Exercise-1 of where the file is located.

You shouldn't have to download additional files.

zenetio, 2017-07-05:
Note that I am not talking about tabletop.pcd but table_scene_lms400.pcd

kylesf, 2017-07-05:
Can you provide the link to where you downloaded the file? I will add it to the classroom :)

kylesf, 2017-07-05:
Classroom is updated.

closed 2017-07-05 by kylesf