udacity/robotics-beta-test-issues #546: Clustering with PCL

yuvaramsingh94, 2017-07-06:
i am experiencing trouble in loading the urdf fie into rviz . these are the steps i followed 

download the repository as mentioned in the material
moved it to catkin_ws/src
run catkin_make at ~/catkin_ws(finished with no error)
set the GAZEBO_MODEL
sourced the workspace (added to bashrc)
run roslaunch sensor_stick robot_spawn.launch

and i am getting this result as in the image

yuvaramsingh94, 2017-07-06:
https://drive.google.com/file/d/0BzmnMmsPbJkVU3ZaZlEySGcyWE0/view?usp=sharing

sahiljuneja, 2017-07-06:
I just tried the above from scratch in the latest VM. It, for some reason, takes time to load up. I am not entirely sure why. So let it run for a bit and slowly it should all load up. 

Also, make sure the path for gazebo model is entirely correct. That's quite important.

closed 2017-07-06 by sahiljuneja