udacity/robotics-beta-test-issues #547: akhil - L2.8. setting up python-cpl

AkhilSharma1, 2017-07-06:
When following the instructions in RoboND-Perception-Exercises github repo, `python setup.py build` gives me 'ModuleNotFoundError: No module named 'Cython' . Adding sudo seems to work here.
I am doing this in RoboNDVM 2.1.0

sahiljuneja, 2017-07-06:
That's interesting. I didn't face such a problem when I installed it yesterday. Will wait to see if more people face this problem or not. Thanks for reporting!

AkhilSharma1, 2017-07-06:
it is L3.8 and not L2.8. Soory :)

AkhilSharma1, 2017-07-06:
I am also using Virtualbox instead of VmWarePlayer in case it matters
[image]

sahiljuneja, 2017-07-06:
You did install Cython first, right? There's a step for that as well. [Just to be sure]

AkhilSharma1, 2017-07-06:
Yes, I ran the command to install it but it says that it is already installed but with some cache related warnings:
[image]

I had to use sudo in a couple of places including when trying to start pcl_viewer

closed 2017-11-21 by kylesf