udacity/robotics-beta-test-issues #550: crl Clustering L04E08

zenetio, 2017-07-07:
I was not able to get the table+objects after launch sensor_stick. I run Windows 10 and I used the last 2 versions of VM. I only get the support and the camera.

https://classroom.udacity.com/nanodegrees/nd209-beta/parts/571766f0-9631-484d-a69f-94ea58110725/modules/e5bfcfbd-3f7d-43fe-8248-0c65d910345a/lessons/2cc29bbd-5c51-4c3e-b238-1282e4f24f42/concepts/02428d63-6f79-40dc-8105-31eda8e0def4

zenetio, 2017-07-07:
OK, after remove and add the PointCloud2 again, now I can see the complete image.

kylesf, 2017-07-07:
Is this matter resolved?

zenetio, 2017-07-07:
Well, this is the way I found to see the image displayed.

kylesf, 2017-07-07:
Okay I will look into it. I have tested both exercises with a fresh copy of the VM and had no problems.

kylesf, 2017-07-20:
Please try with the newest VM and let me know if it is still an issue. Thanks!

zenetio, 2017-07-21:
Yes, it is working in the new VM

closed 2017-07-21 by kylesf