udacity/robotics-beta-test-issues #557: Possible Enhancement: Consider additional walkthrough for P2 for end of course

sahiljuneja, 2017-07-19:
This walkthrough could be [Harsh] talking about the process of setting up the environment and everything before coding the IK_server so that students can emulate projects of their own too. 

Just a consideration, putting it as backlog for now.

kylesf, 2017-07-25:
Done :P

sahiljuneja, 2017-07-25:
No, this is different. This is to help them setup the environment in ROS as in the cafe model, adding the robot themselves, set it up for moveit!, having the entire setup from scratch before writing out the code so that they are familiar with the process. This project is probably the only one where they will be able to do so and we can provide a guide since this doesn't use any specialized tools/frameworks (like Unity or custom pcl etc).

kylesf, 2017-07-25:
Sounds good.

closed 2018-05-03 by jessluls