udacity/robotics-beta-test-issues #560: OpenCV Calibration Notebook

letyrodridc, 2017-07-22:
The notebook throws an error

kylesf, 2017-07-25:
Can you share the specs of the system you are using?

letyrodridc, 2017-07-29:
Sorry Kyle, that was a really short description. 

I think I've tested on my computer instead of the VM. 

My computer has the RoboND anaconda environment for the first Beta-test in Search and Sample return project. I've installed it jupyter notebook.

I use a notebook, Lenovo Yoga 510. 

Processor : Intel® Core™ i5-6200U CPU @ 2.30GHz × 4 
Memory : 7,7 GiB 
Graphics: Intel® HD Graphics 520 (Skylake GT2) 

Runs a Linux Ubuntu 16.04 LTS 64-bit.

kylesf, 2017-07-31:
@ryan-keenan Any thoughts on this?

kylesf, 2017-08-02:
I am unsure why your computer is throwing that error. I have just checked to make sure it is working in the course provided VM. Outside of the VM I am unable to diagnose problems! Closing this issue.

closed 2017-08-02 by kylesf

letyrodridc, 2017-08-02:
Ok I will do it on the VM. thanks Kyle