udacity/robotics-beta-test-issues #562: crl Object Recognition: L05E07

zenetio, 2017-07-25:
I think the Feature Vector graphic should be explained. I mean, who is who in the picture.

https://classroom.udacity.com/nanodegrees/nd209-beta/parts/571766f0-9631-484d-a69f-94ea58110725/modules/a352053a-c3f0-411e-9061-44fb82f38a78/lessons/81e87a26-bd41-4d30-bc8b-e747312102c6/concepts/b1864a2b-11b4-4dfd-a8db-552209f1a38c

kylesf, 2017-07-25:
I am not sure I follow. Do you mean label the axis for this image?
[image]

zenetio, 2017-07-26:
No, I am talking about the Feature Vector graphic resulting from the test. It seems that we have features in 10%, 30%, 60% and so on. Is that correct? But what features? I think that we need an explanation text here.

ryan-keenan, 2017-07-26:
Thanks for this suggestion!  I added an image and some text explanation to this concept.  Please let me know what you think (be sure to refresh the page ;):  https://classroom.udacity.com/nanodegrees/nd209-beta/parts/571766f0-9631-484d-a69f-94ea58110725/modules/a352053a-c3f0-411e-9061-44fb82f38a78/lessons/81e87a26-bd41-4d30-bc8b-e747312102c6/concepts/b1864a2b-11b4-4dfd-a8db-552209f1a38c?contentVersion=1.0.0&contentLocale=en-us

closed 2017-07-26 by ryan-keenan

zenetio, 2017-07-26:
Yes, its is much better! Now we know the meaning of bar distribution in the graph.