udacity/robotics-beta-test-issues #564: crl Project 3 - RViz

zenetio, 2017-07-27:
After launch, in RViz I only see RP2. No map or welcome message.

hvpandya, 2017-07-27:
Can you attach a screenshot of your rviz window, along with any error messages you see in the terminal?

zenetio, 2017-07-27:
It sems to be relate do move_group
[image]

hvpandya, 2017-07-27:
Have you tried restarting after waiting a minute or so? The error message says that the robot could not connect to the move_group action _server, meaning the moveit pipeline did not come up in the time it should have.

I am quite confident this should get resolved by restarting the project. If that does not work, try restarting your VM once. Let me know about your findings.

zenetio, 2017-07-27:
I restarted the project and restarted the VM and still getting the same result. Then I tried with the latest VM and now I dont have Rviz and getting the error
[ERROR] [1501183167.964119894, 834.082000000]: couldn't resolve publisher host []

closed 2017-11-21 by kylesf