udacity/robotics-beta-test-issues #565: pcl_viewer shortcuts could be add

letyrodridc, 2017-07-27:
Suggestion: In the exercise 1, some information about how to rotate the view or zoom in pcl_viewer could be useful to students to verify the cloud points.
For example, using plus sign (+) you can zoom in or use the mouse to rotate the scene. 
Link to classroom:  https://classroom.udacity.com/nanodegrees/nd209-beta/parts/571766f0-9631-484d-a69f-94ea58110725/modules/e5bfcfbd-3f7d-43fe-8248-0c65d910345a/lessons/8d51e0bf-0fa1-49a7-bd45-e062c4a2121f/concepts/02cbb56e-9e54-4c08-977b-df149cb0bca4

sahiljuneja, 2017-07-27:
You mean something different than what was covered here - https://classroom.udacity.com/nanodegrees/nd209/parts/7b2fd2d7-e181-401e-977a-6158c77bf816/modules/8855de3f-2897-46c3-a805-628b5ecf045b/lessons/91d017b1-4493-4522-ad52-04a74a01094c/concepts/809ef993-dbbb-49d4-8520-0bae34921a60 ?

jchernus, 2017-07-29:
Or maybe even a "Reminder: if you've forgotten how to... please visit the lesson on Rviz here: <link>"

letyrodridc, 2017-07-29:
Hi  @sahiljuneja @jchernus 
Maybe I'm missing something but I'm not talking about Rviz. I'm referring to the "pcl_viewer" that we use for perception exercises. 

Maybe "pcl_viewer" is implemented over Rviz(???) The issue is that the student doesn't know that can zoom in and out, rotate, to see the cloud points. Eventually, is going to find it out, but why don't suggest to zoom in to see better? Pressing + (I think that was the shortcut). 

Some more information about the shortcut there I think it could help. Thanks!

sahiljuneja, 2017-07-30:
Oh ok. I will look into this and see what can be added.

closed 2018-05-25 by sahiljuneja