udacity/robotics-beta-test-issues #566: Update VM keyboard selection

kylesf, 2017-07-28:
It would help us here in Brazil to be able to choose the Portuguese (ABNT2) layout!

georgebrindeiro, 2017-07-29:
As per @tokyo_adam's suggestion in Slack, a good alternative to always updating the VM keyboard selection might be to add the following instructions in the classroom:

"Right click the panel -> Add / Remove Panel Items -> Add -> Keyboard Layout Handler

That adds an icon to the panel, and by right clicking it and selecting "Settings", a GUI tool for managing keyboard layouts shows up.

https://askubuntu.com/questions/788244/how-to-change-keyboard-layout-in-lubuntu-15-10"

I can confirm I could easily add my keyboard layout using that method. Maybe adding the Keyboard Layout Handler to the VM is helpful enough.

kylesf, 2017-08-14:
# Adding more keyboard feedback

May I kindly ask you to add the Swiss German keyboard layout in the next version. Thank you. In the meantime I will follow @tokyo_adam 's advice above.

For correct recognition of Logitech Unifying devices, it might be helpful to add `solaar` (https://pwr.github.io/Solaar/)

kylesf, 2018-03-12:
This update is being held and we will be moving to workspaces. Therefore the VM will be deprecated. Thanks for submitting!

closed 2018-03-12 by kylesf