udacity/robotics-beta-test-issues #567: crl Object Recognition L05E17

zenetio, 2017-07-28:
When I run the object_recognition script I get cloud_out() undefined. Where is this function defined?

https://classroom.udacity.com/nanodegrees/nd209-beta/parts/571766f0-9631-484d-a69f-94ea58110725/modules/a352053a-c3f0-411e-9061-44fb82f38a78/lessons/81e87a26-bd41-4d30-bc8b-e747312102c6/concepts/8389976c-550a-4fde-a530-65ca24b31f05

hvpandya, 2017-07-28:
cloud_out is not a function but a point cloud that you created in EX-2. Have you completed ex-2? If so, you will need that code as a part of your solution for EX-3.

jchernus, 2017-07-30:
Harsh, in Ex2 we had a could_objects and cloud_table point clouds. I think the confusion is simply with the variable name.

sahiljuneja, 2017-08-07:
Updated the name and comment on what it refers to. That should allow students to relate it to Ex-2 better.

closed 2017-08-07 by sahiljuneja