udacity/robotics-beta-test-issues #570: Perception - Exercise 1 - Outlier Removal Filter ROS to PCL and variable pcl_msg

letyrodridc, 2017-07-29:
In file `template.py` it says:
```
def pcl_callback(pcl_msg):

    # TODO: Convert ROS msg to PCL data
```

I don't understand. I think there is a contradiction between variable name `pcl_message` and starting comments ROS msg to PCL_data.

Is the method input in ROS msg? or is it a PCL data? 

If it's a ROS msg, the variable should call `ros_message` instead of `pcl_message`.

If it's a PCL msg, the comment should convert PCL to ROS.

jchernus, 2017-07-29:
We're receiving and transmitting all data via ROS messages, but to be able to work with it - we need as PCL. Thus, first we convert ROS to PCL, and after processing the data, we convert back to a ROS message to publish it on another topic.

letyrodridc, 2017-07-29:
Yes. But the function parameter name is pcl_message when it is a ros_message

sahiljuneja, 2017-07-30:
I understand your point and why it could potentially cause confusion. I am currently unsure of all the changes that would be required in context of this, and if it's actually necessary or not in the long run. Because it was probably meant to indicate a ROS "msg" which has point cloud data so the "pcl" is to indicate that I believe, while "msg" refers to ROS' message description. But I can understand why a better variable name could help and avoid potential confusion.

So I will put this into backlog and we can return to it later. Thanks!

cc - @hvpandya if you agree or not, let us know.

hvpandya, 2017-07-31:
It is just normal convention to append "_msg" at the end of ros messages, this is something students will have to learn getting used to. Great explanations @jchernus  and @sahiljuneja. 

Closing this as it is not a bug.

closed 2017-07-31 by hvpandya