udacity/robotics-beta-test-issues #574: 3D perception project error on gazebo and rviz

yuvaramsingh94, 2017-07-30:
https://drive.google.com/open?id=0BzmnMmsPbJkVdUV4d2lGc1pKM0k

this following error happens when i run the demo cmd provided in the course material after setting up the environment for the project . 
hitting next also has no effect on the rviz , no marker displayed (title) . not sure of the issue .

this is my setup 
i have a catkin_ws workspace where i have the 2nd and sensor_stick project 
i created a new workspace and named it catkin_ws2 and cloned the project . had no error during catkin_make .

kylesf, 2017-07-31:
Are you using the supplied VM?

sahiljuneja, 2017-07-31:
This is the current solution thanks to a beta-tester. 
```
I actually could solve this demo issue. Here is what I did:

1. I ran these commands on two commands on two different terminals:
`roslaunch pr2_robot pick_place_demo.launch`
`rosrun moveit_ros_move_group move_group`

2. I ensured I had `ros_moveit` installed.
In case you don't have `ros_moveit`, make sure to install it using `sudo apt-get install ros-kinetic-moveit`

3. This solved my demo issues. I'm not sure if it'll work for you guys, but worth trying.

I think running the two commands `roslaunch pr2_robot pick_place_demo.launch`
`rosrun moveit_ros_move_group move_group` separately in two terminals permits us to run `move_group` independently. So it doesn't die and the demo project which uses it works well now.

Not sure if I'm totally right though, but at least the demo is now working well for me (
```

While the above seems to be a temporary fix, Harsh is working on a script to help solve the problem.

closed 2017-07-31 by sahiljuneja