udacity/robotics-beta-test-issues #575: VM: Auto-source in terminal not working

jchernus, 2017-07-30:
As evident below:

Do you want to source ROS in this workspace (y/n): y
ROS sourced!
robond@udacity:~/catkin_ws/src/sensor_stick$ roscd sensor_stick
roscd: No such package/stack 'sensor_stick'
robond@udacity:~/catkin_ws/src/sensor_stick$ cd ~/catkin_ws/
robond@udacity:~/catkin_ws$ source devel/setup.bash 
robond@udacity:~/catkin_ws$ roscd sensor_stick
robond@udacity:~/catkin_ws/src/sensor_stick$

sahiljuneja, 2017-07-30:
That option is to select between ROS and the Anaconda Environment. In `.bashrc` you will still have to uncomment the `source devel...` line to have auto-sourcing setup.

kylesf, 2017-07-31:
Sahil is correct in this matter! Closing!

closed 2017-07-31 by kylesf