udacity/robotics-beta-test-issues #578: Perception - Exercise 3 - Train your SVM Solution Quiz incorrect

letyrodridc, 2017-07-30:


jchernus, 2017-07-30:
http://imgur.com/a/BluGC

Represents how the correct values were obtained. The confusion matrix doesn't show only 'correct' or 'incorrect' classifications. It shows ALL classifications. The values on the diagonal would be considered correct, as the Predicted Label matches the True Label. Everything off the diagonal is incorrect. 

Thus, when question 2 asks about a correct classification, we are looking at the diagonal. And question 3 asks about a misclassification, so we are looking at where "True Label: Create" matches up with "Predicted Label: Beer".

Does this make any more sense now?

letyrodridc, 2017-07-30:
ahh yes, anything that is not in the diagonal is misclassification. Go it. Thanks Julia

closed 2017-07-30 by letyrodridc