udacity/robotics-beta-test-issues #587: introduction to controls(using dt for integral part)

yuvaramsingh94, 2017-08-09:
a typical pid controller will have dt for I and D term. here we are using dt for only I and i was able t get away with not using dt for D term (my submission was accepted ). 

can you explain the reason :)

kylesf, 2017-08-09:
Could you let me know more information so I can replicate this error and fix it?

yuvaramsingh94, 2017-08-09:
in artificial intelligence for robotics under pid implementation section 

def run(param1, param2, param3):
    myrobot = robot()
    myrobot.set(0.0, 1.0, 0.0)
    speed = 1.0 # motion distance is equal to speed (we assume time = 1)
    N = 100
    myrobot.set_steering_drift(10.0 / 180.0 * pi) # 10 degree bias, this will be added in by the move function, you do not need to add it below!
    #
    # Enter code here
    #
    old_error = myrobot.y
    total = 0.0
    for i in range (0,N):
        error = myrobot.y
        diff = error - old_error
        total +=error
        old_error = error
        steer = -(param1*error)-(param2*diff)-(param3*total)
        myrobot= myrobot.move(steer,speed)
        print myrobot,steer

# Call your function with parameters of (0.2, 3.0, and 0.004)
run(0.2, 3.0, 0.004)

here we are not using any dt term because we have removed the dt term from the entire equation 

https://www.youtube.com/watch?v=vnOcv2xkROY
here is the course from coursera where they discussed about delta t

kylesf, 2017-08-09:
That is for the artificial intelligence for robotics course. 

In the PID quiz for our beta section you can see in this snippet that dt is taken into account
```
        # Sum the errors
        self.error_sum_ += error * delta_time
        
        # Calculate the delta_error
        ########################################
        delta_error = error - self.last_error_
        ########################################
        
        # Update the past error with the current error
        ########################################
        self.last_error_ = error
        ########################################
        
        # Proportional error
        p = self.kp_ * error
       
        # Integral error
        i = self.ki_ * self.error_sum_
       
        # Calculate the derivative error here. Be sure to access the 
        # the internal Kd class variable
        ########################################
        d = self.kd_ * (delta_error / delta_time)
        ########################################
```

Now are you saying that you were able to by pass our quiz in the Building a PID controller section?

yuvaramsingh94, 2017-08-09:
i was able t opass with this code . here i have not used delta_time for d calculation     

    delta_error = error - self.last_error_

########################################
        d = self.ki_ * delta_error
        ########################################
        
        # Set the control effort
        # u is the sum of all your errors. In this case it is just 
        # the proportional error.
        ########################################
        u = p+i+d
        ########################################

kylesf, 2017-08-09:
Ahh seems you have beat the test case! Let me make it more strict :)

yuvaramsingh94, 2017-08-09:
:)

yuvaramsingh94, 2017-08-09:
i think i was able to beat it because the dt was a constant value . dt will be effective if the time gap is varying

kylesf, 2017-08-09:
Should be fixed now.

kylesf, 2017-08-09:
Let me know!

kylesf, 2017-08-10:
?

closed 2017-08-12 by kylesf