udacity/robotics-beta-test-issues #593: Controls Project: Drone simulator exits right after launch on the VM with unity installed.

evindj, 2017-08-11:


kylesf, 2017-08-12:
What is you host OS?

evindj, 2017-08-13:
Windows

kylesf, 2017-08-13:
Are you running the windows simulator along side with the linux VM?

evindj, 2017-08-13:
I was running the simulator inside the VM when this happened. The simulator
on the host machine and it launched correctly but I think it could not
communicate with ROS(on the VM)

kylesf, 2017-08-13:
You cannot run the simulator in the VM. It has to be ran on your host computer.

evindj, 2017-08-13:
Is there another configuration that has to be done so the simulator can
communicate with the VM? When I run the simulator in the host and run
roslaunch quad_controller hover_controller.launch I get error Service
[/quad_rotor/x-force_constrained] is not available.

kylesf, 2017-08-13:
Check out the bottom of this README. https://github.com/udacity/RoboND-Controls-Lab

evindj, 2017-08-13:
> Check out the bottom of this README. https://github.com/udacity/
> RoboND-Controls-Lab

Awesome!!! will try it out.

evindj, 2017-08-13:
Just teste it. The simulator is able to connect to roscore but now
run roslaunch quad_controller hover_controller.launch is not able to
communicate with the services on the host. I am trying different network
configuration on the VM will update you on how it goes.

2017-08-13 10:55 GMT-04:00 Innocent Djiofack <djiofack007@gmail.com>:

> Awesome!!! will try it out.
>
>
>> Check out the bottom of this README. https://github.com/udacity/Rob
>> oND-Controls-Lab
>>
>>
>
>
>
>
> *DJIOFACK INNOCENT*
> *"Be better than the day before!" -*
> *+1 404 751 8024 <(404)%20751-8024>*

evindj, 2017-08-15:
I was not able to make this work correctly below is the screenshot of what I got.  Here is my configuration of simulator
{
	"vm-ip" : "192.168.182.129",
	"vm-port" : 11311,
	"vm-override" : true,
	"host-ip": "0.0.0.0",
	"host-override" : false
}
[image]

celibertojr, 2017-08-15:
I did the same mistake. I think would be good to write in capital letters to avoid the same mistake by the students.

evindj, 2017-08-15:
@celibertojr what mistake did you do ? how did you solved it? what should be written in capital letters?

kylesf, 2017-08-15:
I have made it a bold statement in the README. As for @evindj, please DM on slack and we can get you a solution.

closed 2017-11-21 by kylesf