udacity/robotics-beta-test-issues #595: Introduction to Controls - Getting to know your Simulator

cornzyblack, 2017-08-12:
I receive some error messages when trying to test run the default code. Is this normal?

cornzyblack, 2017-08-12:
`Traceback (most recent call last):
  File "vm_main3.py", line 47, in <module>
    import main
  File "/tmp/vmuser_eplqjiqolx/main.py", line 2, in <module>
    import studentMain
  File "/tmp/vmuser_eplqjiqolx/studentMain.py", line 63, in <module>
    import hover_plot
  File "/tmp/vmuser_eplqjiqolx/hover_plot.py", line 71, in <module>
    ax3.plot(time, controller.effort_applied, label='effort', linewidth=3, color = 'red')
  File "/usr/local/lib/python3.4/dist-packages/matplotlib/__init__.py", line 1892, in inner
    return func(ax, *args, **kwargs)
  File "/usr/local/lib/python3.4/dist-packages/matplotlib/axes/_axes.py", line 1406, in plot
    for line in self._get_lines(*args, **kwargs):
  File "/usr/local/lib/python3.4/dist-packages/matplotlib/axes/_base.py", line 407, in _grab_next_args
    for seg in self._plot_args(remaining, kwargs):
  File "/usr/local/lib/python3.4/dist-packages/matplotlib/axes/_base.py", line 385, in _plot_args
    x, y = self._xy_from_xy(x, y)
  File "/usr/local/lib/python3.4/dist-packages/matplotlib/axes/_base.py", line 244, in _xy_from_xy
    "have shapes {} and {}".format(x.shape, y.shape))
ValueError: x and y must have same first dimension, but have shapes (500,) and (501,)`

[image]

kylesf, 2017-08-12:
Try refreshing the browser and resetting the quiz. Please :)

cornzyblack, 2017-08-12:
I've done both of that but get the same error. (I use Chrome)

kylesf, 2017-08-12:
Hmm, let me test with my student account in chrome.

kylesf, 2017-08-12:
Squashed the bug! Thanks for reporting. Let me know if it works for you now.

cornzyblack, 2017-08-12:
Ok, I just refreshed now and tried it again, the error is gone now :)

kylesf, 2017-08-12:
Cool. Thanks again!

closed 2017-08-12 by kylesf