udacity/robotics-beta-test-issues #613: Introduction to Controls: Building a PI controller

alvarovillena, 2017-08-16:
According to the table found in the [PID Control Summary section](https://classroom.udacity.com/nanodegrees/nd209-beta/parts/fc3e1236-996e-4fb8-b317-479ce23c0e1d/modules/95f92697-f568-4063-b6d4-b134503ec984/lessons/6c738150-6af4-4c12-a21d-8bc6ea55d72e/concepts/78c80760-fddc-4807-a72c-79685edf8ec8), increasing the value of the integral constant decreases rise time and steady state error, and increases overshoot and settling time. However, this is not the correct answer for the third question on the [Building a PI controller section](https://classroom.udacity.com/nanodegrees/nd209-beta/parts/fc3e1236-996e-4fb8-b317-479ce23c0e1d/modules/95f92697-f568-4063-b6d4-b134503ec984/lessons/6c738150-6af4-4c12-a21d-8bc6ea55d72e/concepts/b7d754ca-e890-4e41-8198-63857af6b81d).

kylesf, 2017-08-16:
Interesting. Those are the answers in the quiz. What do you get as a response?

alvarovillena, 2017-08-16:
The response is: `Try Again. Check to see what happens when you make the change in the code above!`

thomas-at-em, 2017-08-16:
Can confirm. This quiz is broken but quiz create does have the correct answers. This must be a bug on Udacity backend.

kylesf, 2017-08-16:
Fixed.

closed 2017-08-16 by kylesf