udacity/robotics-beta-test-issues #619: crl Quadrotor Control: L02E10

zenetio, 2017-08-17:
I am getting this output when I launch the hover scenario. It seems that some services cannot be found. 

process[rqt_reconfigure-3]: started with pid [119391]
ERROR: Service [/quad_rotor/camera_pose_type] is not available.
ERROR: Service [/quad_rotor/gravity] is not available.
ERROR: Service [/quad_rotor/x_force_constrained] is not available.
[INFO] [1503000320.986371]: Reconfigure Request: 6.0,            3.0, 2.0, 6.0
ERROR: Service [/quad_rotor/y_force_constrained] is not available.
ERROR: Service [/quad_rotor/x_torque_constrained] is not available.
ERROR: Service [/quad_rotor/y_torque_constrained] is not available.
ERROR: Service [/quad_rotor/z_torque_constrained] is not available.
[setup_2d_world-1] process has died [pid 119388, exit code 2, cmd /home/robond/drone_ws/src/RoboND-Controls-Lab/quad_controller/scripts/setup_2d_hover_world.sh __name:=setup_2d_world __log:=/home/robond/.ros/log/536b61ce-8386-11e7-b6b2-000c299f721c/setup_2d_world-1.log].

thomas-at-em, 2017-08-17:
Hmmm, I am going to need more information from you before I can troubleshoot this.
(1) In the lower left corner of the Unity Window, do you see a green circle next to the words "Connected to ROS"?
(2) What OS are you using?
(3) Are you running ROS in a VM?
(4) Are  you running Unity natively on your host OS?

zenetio, 2017-08-18:
Ok, I deleted my repo, downloaded a new one and now the error message was fixed. Any way, the sim still doesn't move. I saw one student in the forum with the same problem. I have tried to figure out where is the problem but no luck until now. When I move the sim manually on Windows, I can see the reflected values echoed on ROS. If I send a manual camera command from ROS I can see it reflected on sim. I can send set_pose command and sim reflects the command. But roslaunch or rosrun set the sim scene to start running but nothing happens with sim. I am guessing if there is a service or a topic not being handled correctly
1. ROS is connected
2. Windows 10
3. using VM 
4. Unity natively on Windows

zenetio, 2017-08-18:
More comments.
When I roslaunch hover_controller, I can see that sim is subscribed in /quad_rotor/cmd_force and when I move kb knob, I can see the z value changing in this topic. But when I the telemetry panel in sim, there is no change in z values. By the other hand, if I send a ROS pose command manually, I can see the value reflected in the sim telemetry panel. That said, it seems there is a problem with cmd_force in sim side once I see the values changing in ROS topic but not in sim telemetry panel

zenetio, 2017-08-18:
More comments.
Ok, checking the sim log file on Windows it shows an error with cmd_force which lead us to see that there is an issue impacting the sim movement. Please check the error below.

Failed to contact publisher [udacity:44227] for topic [/quad_rotor/cmd_force]
(Filename: C:/buildslave/unity/build/artifacts/generated/common/runtime/DebugBindings.gen.cpp Line: 51)
Unknown error while handling XmlRpc call to pubUpdate

zenetio, 2017-08-19:
OK, I just checked and it works with VM 2.1.0. Any tip what I need update in VM 2.0.1 to fix the issue? I have many things installed on VM 2.0.1 and would like to continue to use it.

kylesf, 2017-08-19:
Yes add `export ROS_IP=`echo $(hostname -I)` to your bash profile :)

closed 2017-08-28 by kylesf