udacity/robotics-beta-test-issues #624: crl Quadrotor Control: L02E10

zenetio, 2017-08-20:
After update bashrc file with ROS_IP, I could have hover working couple times on VM 2.0.1 and 2.1.0. Then for some reason, it stopped to work on both VMs. Then I tried attitude but the result was the same: not working as well. The drone connection is OK and I dont see any error in drone log file. I removed devel and build directories and ran catkin_make but no success. It seems that something screwed in the VM. I am guessing if there is something wrong with nodes. Below you can see the output of rosnode list command. Note that for cmd_force the field name is empty, where I thing should be equal to pose topic

robond@udacity:~$ rosnode info /hover_controller
--------------------------------------------------------------------------------
Node [/hover_controller]
Publications: 
 * /hover_controller/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /rosout [rosgraph_msgs/Log]
 * /hover_controller/parameter_updates [dynamic_reconfigure/Config]
 * /quad_rotor/cmd_force [geometry_msgs/Wrench]

Subscriptions: 
 * /quad_rotor/pose [geometry_msgs/PoseStamped]

Services: 
 * /hover_controller/set_logger_level
 * /hover_controller/set_parameters
 * /hover_controller/get_loggers

contacting node http://192.168.132.170:35895/ ...
Pid: 11628
Connections:
 * topic: /hover_controller/parameter_descriptions
    * to: /rqt_reconfigure
    * direction: outbound
    * transport: TCPROS
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS
 * topic: /hover_controller/parameter_updates
    * to: /rqt_reconfigure
    * direction: outbound
    * transport: TCPROS
 * topic: /quad_rotor/cmd_force
    * to: /
    * direction: outbound
    * transport: TCPROS
 * topic: /quad_rotor/pose
    * to: / (http://SAMSUNG-PC:57332/)
    * direction: inbound
    * transport: TCPROS

kylesf, 2017-08-28:
Can you tell me some more about the system you are using?

kylesf, 2017-08-31:
Update?

zenetio, 2017-09-01:
VMware 12 Pro with VM 2.0.1 or 2.1.0 running on Windows 10.

kylesf, 2017-09-01:
PM me on slack and we can troubleshoot your issue.

closed 2017-09-04 by kylesf