udacity/robotics-beta-test-issues #641: Semantic Segmentation: The link going to the "classroom" not accessible to beta students

Shreeyak, 2017-09-22:
Issue on this page: https://classroom.udacity.com/nanodegrees/nd209-beta/parts/4405411f-bdc8-43fb-9e2b-b4b98a61c760/modules/375d6b1e-b31a-4d2e-b33a-996167faf77e/lessons/99db3ae7-9e87-40ba-b83c-5d1428643d2d/concepts/93bb618f-bbac-4abf-82cc-49965ac53fc8

The link (given in multiple places across the DL section) is not accessible to beta students. Please do fix the others too.

sahiljuneja, 2017-09-22:
Yes, those are intentional. Will be updating them soon. Thanks!

kylesf, 2017-09-25:
Closing this until we have official launch. Will then walk through the content making sure links are fixed. Just fyi. Classroom links will layaways be broken for beta testers :(

closed 2017-09-25 by kylesf