udacity/robotics-beta-test-issues #642: L06 Segmentation: RoboND-DeepLearning readme typo

Shreeyak, 2017-09-23:
In the readme for this project: https://github.com/udacity/RoboND-DeepLearning.git,

One of the dependencies is stated as OpenCV 2, which I think is an error. Even though the package is named as cv2, it is actually OpenCV 3. Current stable versions are 3.2.0 and 3.3.0. 

Also, maybe you should rename this repo, as it sounds like the final project name. Following previous conventions might be better, like RoboND-SegmentationLab or similar.

Shreeyak, 2017-09-23:
2. "Download the training dataset from above and extract to the project data directory" . Students would need to extract the validation set too.

Shreeyak, 2017-09-23:
3. "Training CNNs is a very compute-intensive process. If your system does not have a recent Nvidia graphics card, with *cuDNN* installed " -> should be CUDA, or CUDA+cuDNN

Shreeyak, 2017-09-23:
4. "in data_iterator_template.py" -> should be utils/data_iterator_template.py . I initially thought I might have accidentally deleted that file, when I did not see it in the code folder.

kylesf, 2017-09-25:
1) Done on our side so it will be fixed in next public release. As for repo name we will keep that in mind.

2) I think these may be reworded later but it is in that sentence focusing on just the use of training data

3) Fixed for nest release. 

4) This part will be depreciated.

closed 2017-09-25 by kylesf