udacity/robotics-beta-test-issues #643: L06 Segmentation: RoboND-Deeplearning ReadMe: dependencies might need to be installed separately

Shreeyak, 2017-09-23:
I have Anaconda setup with python 3.6, jupyter and tensorflow. However, there were still some packages that I needed to install, like socketio and eventlet. It will be nice to provide instructions on how to install these dependecies.
Eg: after activating env, `conda install python-socketio` and `pip install eventlet`

Shreeyak, 2017-09-23:
Providing a requirements file to be run with pip would be a good way to go about it.

Shreeyak, 2017-09-23:
Also, my conda installation comes with pillow, a "friendly PIL fork". So that might be another cause for confusion.

kylesf, 2017-09-25:
Did you install the supplied robond env? If not I think dependencies are covered on the github README?

closed 2017-11-21 by kylesf