udacity/robotics-beta-test-issues #646: Project implementation: feelings of a student

yannrs, 2017-09-25:
With the point of view of a student, all functions preimplemented are very fuzzy, as there are few comments, and they didn't implemented neither of them.
I would advise to let students implement some parts of these fonctions, with some strong guidiance
The fuzziest is the SeparableConv2DKeras.

yannrs, 2017-09-25:
I didn't saw the new part on the lesson semantic segmentation, which solve my worries

yannrs, 2017-09-25:
Lesson6, 8.Lab: Encoder: I would recommend to add a figure which illustrate the steps use with the separavbleConv2D compared to commonly used conv2D.
With a visual point of view it should be clearer for a wider range of students

kylesf, 2017-09-25:
Good idea! Will add it to back log

kylesf, 2017-11-21:
@sahiljuneja Thoughts on this?

closed 2018-05-03 by danainschool