udacity/robotics-beta-test-issues #647: Project implementation: import model

yannrs, 2017-09-25:
* To load the model, I would suggest this approach:

from keras.models import model_from_json
# load json and create model
model = model_from_json(json.load(open(save_dir+"model.json", "r")))
# load weights into new model
model.load_weights(save_dir+"model_weights.h5")

kylesf, 2017-09-25:
Why do you suggest this approach?

yannrs, 2017-09-25:
Because with this one, you can load weights model from a different architecture than the one you implemented before.

yannrs, 2017-09-27:
For example, I can't load the example's weights you provided, because I havn't the architecture you used.

kylesf, 2017-09-27:
Those weights are just meant to be used for performance testing with the simulator

yannrs, 2017-09-28:
This was an example, but I can load my own weights.
if I use the function, my variable "model" become None.
I have to load everything with .model_from_json and then .load_weights

closed 2017-11-21 by kylesf