udacity/robotics-beta-test-issues #649: [Windows]Unity simulatior: crash

yannrs, 2017-09-25:
When I launch the unity simulator with the server and the realtime_follower, python crash because of an incompatibility:

could not create cudnn handle: CUDNN_STATUS_NOT_INITIALIZED
2017-09-25 23:42:56.283034: E c:\tf_jenkins\home\workspace\release-win\m\windows-gpu\py\35\tensorflow\stream_executor\cuda\cuda_dnn.cc:366] error retrieving driver version: Unimplemented: kernel reported driver version not implemented on Windows

kylesf, 2017-09-25:
This is possibly an already known issue. If installs have been varified, and tensorflow runs in normal situations without the server.py running, then there is a workaround, and the issue will be fixed in the updated follower code 

The work around is to only start `realtime_follower.py`  after both server.py is running, and the sim is already transmitting data via socket

yannrs, 2017-09-25:
Thanks, it's better;
But now the issue is:
WARNING:socketIO-client:localhost:4567/socket.io [connection error] unexpected status code (400 Bad Request)
disconnect
disconnect
disconnect

bkinman, 2017-09-25:
This a separate and known issue. It will be fixed in the next release! Thanks @yannrs

closed 2017-11-21 by kylesf