udacity/robotics-beta-test-issues #653: Lab Segmentation: Modification to perform for Hyperparameters

yannrs, 2017-09-26:
I would advise to add more comment on the cell hyperparameters, to explain that they mean, even if I'm sure that almost all of them will have already seen this basic parameters.
But for teaching it's always good to repeat. 

Maybe you can also add  more parameters to tune ? And some of them already have a good value.

Also there is still a "TODO" for compiling the model, while it shouldn't

sahiljuneja, 2017-09-26:
Could you specify which parameters you think already have a good value and it might be good to tune?

I will remove the TODO, thanks!

yannrs, 2017-09-26:
For example I would add a parameter to validate the model. more with a ratio point of view. Such that the student also have to take into consideration that this parameters is also important

Shreeyak, 2017-09-30:
i concur. I raised this issue earlier. This is what I would like to see, please correct if wrong:

https://slack-files.com/T4ET9BQ3U-F7CFKRC07-aed847ceb3

sahiljuneja, 2017-10-02:
We recently modified the notebooks to include more tunable hyperparameters. Hopefully, those should help more. Thanks!

closed 2017-10-02 by sahiljuneja