udacity/robotics-beta-test-issues #654: Lab Segmentation: Load weights

yannrs, 2017-09-26:
With the function:
        model = keras.model.load_weights(model_path)

It lead to :
AttributeError: module 'tensorflow.contrib.keras.python.keras.models' has no attribute 'load_weights'

So it should be :

        model = your_model.load_weights(model_path)

kylesf, 2017-09-26:
Try `model = model.load_weights(model_path)`

yannrs, 2017-09-26:
Yes, this is what we should do outside a function, but in this case, I understand that we need to use :

     model = your_model.load_weights(model_path)

The function: 

   def load_weights(your_model, your_weight_filename):
      model_path = os.path.join('..', 'data', 'weights', your_weight_filename)
      if os.path.exists(model_path):
          model = keras.model.load_weights(model_path)
          return model
      else:
          raise ValueError('No weight file found at {}'.format(model_path))

sahiljuneja, 2017-09-26:
Yeah, I think your_model.load_weights is correct based on some testing. I have updated that. Let me know if the above is working as intended for you. Thanks!

closed 2017-09-26 by sahiljuneja