udacity/robotics-beta-test-issues #674: shreeyak - segmentation lab - give default value of batch size

Shreeyak, 2017-09-29:
`steps_per_epoch` and `validation_steps` should be dependent on `batch_size`. Kindly provide a default batch size. 

Also mention a range of values to try out for `learning_rate` and `num_epochs`. I know the point is to experiment, but this model simply takes way to long to train. A broad range would also be fine, like say try a learning rate b/w 0.1 and 0.001.

danzelmo, 2017-09-29:
We have decided to leave as is. This is something that can be experimented with.

Shreeyak, 2017-09-30:
I feel some explanation of these terms is warranted, if students are going
to experiment on their own. Explain in minibatches, what batch_size,
steps_per_epoch do.

As we discussed in Slack, the meaning of an epoch itself has changed:
updates happen at every step instead of every epoch now. This information
should be conveyed, since that now affects the meaning of all these terms
as we've been taught.
 Also good to mention is that in typical applications, they are run for
20-50 epochs. Just as a reference.

danzelmo, 2017-09-30:
Minibatches are covered in lesson 2, as well as a brief intro to stochastic gradient descent. The 20-50 epochs is not always the case. Imagine your dataset had 20million similar examples, then less then one epoch would be enough. We determine how long to train based on validation loss. You can look at the keras model api documentation https://keras.io/models/model/ search for fit_generator for an explanation of each of terms, we could potentially, add a link to this, in the notebook documentation.

Shreeyak, 2017-09-30:
Haha! Exactly, see? The updates happening at the steps changes everything.
I hope you can understand how these parameters might not quite be clear to
the students and why it bears more explanation.

Also, to make this more evident, can the plots be drawn to have multiple
updates per epoch? Say every 10 or 20 or 50 steps, plot is updated, and
drawn together at the end of the epoch. This would demonstrate the concept
well.

danzelmo, 2017-09-30:
Reclosing because this really is covered in the course material.

closed 2017-09-30 by danzelmo