udacity/robotics-beta-test-issues #677: shreeyak - seglab - prediction steps broken

Shreeyak, 2017-09-29:
So running `` gives me an error:  
```
TypeError                                 Traceback (most recent call last)
<ipython-input-24-c4320e95393d> in <module>()
      1 # generate predictions, save in the runs, directory.
      2 run_number = 'run1'
----> 3 gt_folder, pred_foldr = model_tools.write_predictions_grade_set(model,run_number,'validation')

TypeError: write_predictions_grade_set() missing 1 required positional argument: 'grading_dir_name'
```

Tried going through the code, but can't get it.

Shreeyak, 2017-09-29:
Also look at the last cell `scoring_utils.score_run(validation_path, output_path)`. There is no variable called validation_path.

It's like there's been a huge mix up when releasing this. Please test the code before publishing this again.

danzelmo, 2017-09-29:
Updated write_predictions_grade_set, you can close the issue if its fixed.

kylesf, 2017-09-29:
@Shreeyak Is the error resolved?

danzelmo, 2017-09-29:
This issue has been fixed in segmentation repo

closed 2017-09-29 by danzelmo