udacity/robotics-beta-test-issues #679: shreeyak - project follow me: the depth image should have a black background

Shreeyak, 2017-09-29:
In the sim, can't make out the output of the depth camera at all. Most of it is transparent. Can make out its the depth output only in the first few seconds of launching it. Will be much better if a black background can be added. Can visualize the depth camera.

Shreeyak, 2017-09-29:
Can see the depthcam output here:
[image]

After first few seconds, it is focused on nearby objects and output not visible due to transparency.
[image]

demi180, 2017-09-29:
Good catch. Fix should be coming either today or Monday.

kylesf, 2017-11-21:
@danzelmo Whats the status with this?

closed 2018-05-03 by danainschool