udacity/robotics-beta-test-issues #685: Project follow me:  preprocess_ims.py not working, doesn't find path

yannrs, 2017-09-29:
There aren't any errors, but with a lite of debug, I found out, that the issue came from path to folders and images. by removing one '*' it does something

danzelmo, 2017-09-29:
Where are your images stored?

yannrs, 2017-09-29:
the are now on 
data/raw_sim_data/train/run1
data/raw_sim_data/validation/run1

danzelmo, 2017-09-29:
Thats strange. the structure should be  data/raw_sim_data/train/run1/IMG/   then a bunch of images inside. I won't be able to look at this for a couple of hours, but will look at it as soon as I can.

danzelmo, 2017-09-30:
Please reopen if there is some further issue. I have checked and everything appears to be working. 
inside the IMG/ directory there should be 4 types of images. One rgb images and 3 depth images. With names containing cam1_ through cam4_.

closed 2017-09-30 by danzelmo