udacity/robotics-beta-test-issues #687: follow me simulator v1.2.1 crashed

yuvaramsingh94, 2017-09-30:
simulator crash if the spawn timer is set to short 
i am using the windows version 

crash report 
https://drive.google.com/open?id=0BzmnMmsPbJkVR0FqcS10U001RW8

danzelmo, 2017-09-30:
How long did it take for the crash to occur after setting the spawn timer to short setting? 
What graphics mode were you using?

yuvaramsingh94, 2017-09-30:
it takes about 5-10 sec to crash . i am not sure about the graphics setting because i cant find any setting tab to set the graphics mode :(

danzelmo, 2017-09-30:
It usually starts out on good or fantastic graphics settings. press `f5` to switch between settings. On linux I can get between 3-10 minute runs at least on `fastest` graphics mode. I have been having this issue for a week at least but this is the first replication on another machine.

yuvaramsingh94, 2017-09-30:
will try that now :+1:

closed 2017-11-21 by kylesf