udacity/robotics-beta-test-issues #688: shreeyak - seglab and project4 - add tensorboard

Shreeyak, 2017-09-30:
We can use tensorboard to visualize our plots and it takes just 1 line to be changed.

```
#logger_cb = plotting_tools.LoggerPlotter()
logger_cb = keras.callbacks.TensorBoard(log_dir='./logs2', histogram_freq=0, batch_size=32, write_graph=True, write_grads=False, write_images=False, embeddings_freq=0, embeddings_layer_names=None, embeddings_metadata=None)
```

Shreeyak, 2017-09-30:
Here it is working successfully on my machine:
[image]

Shreeyak, 2017-09-30:
Doc here: https://keras.io/callbacks/#tensorboard

sahiljuneja, 2017-10-02:
Thanks. Currently, there are no plans to add tensorboard because that might require additional content as well. I will add this to the backlog for a later consideration.

closed 2018-05-03 by danainschool