udacity/robotics-beta-test-issues #692: Project follow me: preprocess_ims.py not fully working, TypeError

yannrs, 2017-10-01:
I don't know why, but the preprocess_ims.py isn't finishing well. I obtain images on folder Preprocessing_sim_data, but with the error:
(tensorflow11) C:\Users\yann\deepcars-master\RoboND\Project 4 - Follow Me\RoboND-DeepLearning-Project-master\code>python preprocess_ims.py
Traceback (most recent call last):
  File "preprocess_ims.py", line 155, in <module>
    combine_masks(out_val_dir)
  File "preprocess_ims.py", line 110, in combine_masks
    misc.imsave(os.path.join(dir_base, new_fname), im)
  File "C:\ProgramData\Anaconda3\envs\tensorflow11\lib\site-packages\scipy\misc\pilutil.py", line 197, in imsave
    im = toimage(arr, channel_axis=2)
  File "C:\ProgramData\Anaconda3\envs\tensorflow11\lib\site-packages\scipy\misc\pilutil.py", line 343, in toimage
    bytedata = bytescale(data, high=high, low=low, cmin=cmin, cmax=cmax)
  File "C:\ProgramData\Anaconda3\envs\tensorflow11\lib\site-packages\scipy\misc\pilutil.py", line 94, in bytescale
    cscale = cmax - cmin
TypeError: numpy boolean subtract, the `-` operator, is deprecated, use the bitwise_xor, the `^` operator, or the logical_xor function instead.

danzelmo, 2017-10-01:
Did you update to the newest version of the project lab, there was a bug there where a `/` was left in this will cause problems on a windows. machine pull the newest version of the project and see if this problem persists.

yannrs, 2017-10-01:
I have the newest version of the project. But I even just redownloaded the project to be sure

danzelmo, 2017-10-01:
do you have pillow installed? `pip install pillow`

closed 2017-11-21 by kylesf