udacity/robotics-beta-test-issues #710: C++ Rover Freezes after attempting pickup

adammalpass, 2017-10-18:
I am using the Mac 64-bit version of the sim. Communication to VM worked first time with no issues and I could steer the rover around. However after pressing `p` to attempt to pick up the rock the rover freezes. If I press other keys in the terminal the terminal continues to update the new commands, but rover doesn't change.

karimchamaa, 2017-10-18:
Is it freezing forever? or the rover starts moving again after picking up the rock?

adammalpass, 2017-10-18:
Sorry, I should clarify. If a rock is present and I press `p` the the pick operation is executed correctly, and afterwards the rover can continue to be controlled without issues.

However if there is _no_ rock close by and I press `p` then the rover freezes _forever_ and you can't retake control of the rover.

karimchamaa, 2017-10-18:
Thanks for clarifying. I will look for a solution

karimchamaa, 2017-10-20:
Resolved the issue and updated the simulators. Feel free to download it again and test it! Thanks

closed 2017-10-20 by karimchamaa