udacity/robotics-beta-test-issues #718: [Intro C++][11.] Suggestion - Add a quiz with pointer operations

yannrs, 2017-10-21:
Add a quiz, where you can ask the student what will be the output when the input is x+&y, where x and y were defined before

karimchamaa, 2017-10-23:
You mean something like: 
int x=3;
int y=6;
cout << x+&y;
This will give a HEX answer that students will not be able to identify, right?

yannrs, 2017-10-23:
Yes for example

But I also meant something like :
   int  var = 20;   // actual variable declaration.
   int  *ip;        // pointer variable 

   ip = &var;       // store address of var in pointer variable

   cout << var << endl;
   cout << ip << endl;
   cout << *ip << endl;

   var ++;

   cout << var << endl;
   cout << ip << endl;
   cout << *ip << endl;

karimchamaa, 2017-10-23:
Added something similar which is pointer reference to an operation. Thanks for suggesting!

closed 2017-10-23 by karimchamaa