udacity/robotics-beta-test-issues #721: arek - Lesson 5: Fully Convolutional Networks - 8. Transposed Convolution Quiz - Error

867sec, 2017-10-26:
arek - Lesson 5: Fully Convolutional Networks - 8. Transposed Convolution Quiz - Error

After clicking on Submit button for the quiz the pop up message shows up saying "Try Again" with traceback even though using the solution code from the solution tab. Please see screenshot attached.

File "", line 4, in 
  TypeError: 'NoneType' object has no attribute 'getitem'
  Environment: {'executor_result': None, 'grade_result': {'comment': 'Thank you for your submission!', 'feedback': '', 'completed': None, 'test_results': [], 'bin_codes': [], 'correct': False}, 'executor_status': None}
  Submission: {'_operation': 'REGRADE', u'quiz.py': u'import oldtensorflow as tf\nimport numpy as np\n\n\ndef upsample(x):\n    """\n    Apply a two times upsample on x and return the result.\n    :x: 4-Rank Tensor\n    :return: TF Operation\n    """\n    # TODO: Use tf.layers.conv2d_transpose\n    return tf.layers.conv2d_transpose(x, 3, (2, 2), (2, 2))\n\n\nx = tf.constant(np.random.randn(1, 4, 4, 3), dtype=tf.float32)\nconv = upsample(x)\n\nwith tf.Session() as sess:\n    sess.run(tf.global_variables_initializer())\n    result = sess.run(conv)\n\n    print(\'Input Shape: {}\'.format(x.get_shape()))\n    print(\'Output Shape: {}\'.format(result.shape))\n', u'solution.py': u'import oldtensorflow as tf\nimport numpy as np\n\n\ndef upsample(x):\n    """\n    Apply a two times upsample on x and return the result.\n    :x: 4-Rank Tensor\n    :return: TF Operation\n    """\n    # TODO: Use tf.layers.conv2d_transpose\n    return tf.layers.conv2d_transpose(x, 3, (2, 2), (2, 2))\n\n\nx = tf.constant(np.random.randn(1, 4, 4, 3), dtype=tf.float32)\nconv = upsample(x)\n\nwith tf.Session() as sess:\n    sess.run(tf.global_variables_initializer())\n    result = sess.run(conv)\n\n    print(\'Input Shape: {}\'.format(x.get_shape()))\n    print(\'Output Shape: {}\'.format(result.shape))\n'}

sahiljuneja, 2017-10-26:
There's currently an issue with Quizzes that is being looked into. So this is most likely part of that I think. Will update when it's fixed.

sahiljuneja, 2017-10-27:
This should be fixed now!

closed 2017-10-27 by sahiljuneja