udacity/robotics-beta-test-issues #723: Shreeyak Project 4: Follow Me - Rubric points to be amended

Shreeyak, 2017-11-14:
The rubric points to stand out in project 4 follow me are as follows:

To make your project stand out, try one or more of the following:

Recreate the project in TensorFlow without the use of Keras. If successful, compare your results, does code perform the same? Why or why not? Include both your Keras and TensorFlow models for review.

Can you make the vehicle follow another object? What about a hand-off from one person to the next?

Can you have the perform different flight patterns while following the person (circling, figure-8, etc.)

This is obviously written for a case where the students can control the quad on their own and are taught how to. The project in its current state, however, has nothing to do with control of the quad. These points should, therefore, be removed or changed.

Shreeyak, 2017-11-14:
Also, how are reviewers supposed to correct student submissions in tensorflow, having never written it ourselves?

kylesf, 2017-11-21:
Removed for now. thanks

closed 2017-11-21 by kylesf