udacity/robotics-beta-test-issues #724: shreeyak - project 4 follow me - increase project difficulty

Shreeyak, 2017-11-14:
While the earlier score of 45% was very hard, a score of 40% is very easily achieved, using only the provided data set. 

From my understanding, the project is designed to make the student experiment and explore different avenues, leading to self-discovery. I feel the new goal of 40% final score fails that design goal. 

This matter is subjective and some students are facing difficulty in achieving the score of 40%, hence some further investigation might be required. Most of the people barely need to fiddle around with the hyper-parameters and I find this evident in the project reviews I undertake.

Shreeyak, 2017-11-14:
Here is one project where you can see this: https://review.udacity.com/#!/submissions/853175

Do read the write-up of this project

Shreeyak, 2017-11-15:
Here's another project with details on the process he/she followed: 
https://review.udacity.com/#!/submissions/854926

Shreeyak, 2017-11-20:
This student got 40% in a single shot, according to the write-up, without any additional data:
https://review.udacity.com/#!/submissions/865411

Shreeyak, 2017-11-21:
It's interesting to note that some students are finding this challenging. I spoke to someone who's been at it for 2 weeks and hasn't even cross 30% accuracy.

mikeudacity, 2017-11-21:
I think the fact that some students are finding this challenging leads me to believe this is still a good metric. If all students were capable of doing this without issue, I think we would need to revisit it. 

The project has changed since the original inception so I think the metric should stay for now.

closed 2017-11-21 by mikeudacity