udacity/robotics-beta-test-issues #727: shreeyak - kinematics - L02E07 - diagram wrong

Shreeyak, 2017-12-04:
In the video, one frame does not have the y2 drawn in red. Here is the lesson URL:
https://classroom.udacity.com/nanodegrees/nd209-beta/parts/72800973-80cd-4828-81fd-b3390486892b/modules/8855de3f-2897-46c3-a805-628b5ecf045b/lessons/87c52cd9-09ba-4414-bc30-24ae18277d24/concepts/56fa32b9-4a39-4367-8fcc-c520a65426ef

Shreeyak, 2017-12-04:
[image]

jchernus, 2018-02-08:
Hey! I'm not sure what you mean - the screenshot attached, and all of the graphics that I saw in the video seem to have y2 in red.

Shreeyak, 2018-02-08:
This was so long ago, I don't remember what I meant. Sorry. looks like all the labels are there. 

I'll close the issue.

closed 2018-02-08 by Shreeyak