udacity/robotics-beta-test-issues #761: L05E02: Build MCL in C++

zenetio, 2018-02-19:
I just comment out the print statement and got the error below. It seems something wrong with python

main.cpp:230:11: error: expected ‘;’ before string constant
     Print "I am ready for coding the MCL!"
           ^
Traceback (most recent call last):
  File "vm_main.py", line 33, in <module>
    import main
  File "/tmp/vmuser_idssarmdsd/main.py", line 2, in <module>
    import studentMain
  File "/tmp/vmuser_idssarmdsd/studentMain.py", line 4, in <module>
    call("./main.out")
  File "/usr/lib/python2.7/subprocess.py", line 522, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1327, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

karimchamaa, 2018-02-20:
[Uploading Screen Shot 2018-02-20 at 11.48.49 AM.png...]

karimchamaa, 2018-02-20:
I got no error, remember to reset your quiz and system use a python compiler for the C++ code

karimchamaa, 2018-02-20:
[image]

closed 2018-04-09 by karimchamaa