udacity/robotics-beta-test-issues #764: project - where am i - missing links from resources section

Shreeyak, 2018-02-23:
IN the resources section, there are some missing links. 

An abstract:
"Follow the steps outlined in this Project Lesson (insert link), to create your own ROS package, develop a mobile robot model for Gazebo, and integrate the AMCL and Navigation ROS packages for localizing the robot in the provided map."

"Using the Parameter Tuning section (insert link) as the basis, add and tune parameters for your ROS packages to improve your localization results in the map provided. Feel free to explore new parameters using the resources provided earlier (insert link)."

sahiljuneja, 2018-02-23:
Can you provide the link to this section? There's no "resources" section in the Classroom for localization project, I believe.

Shreeyak, 2018-02-23:
Ah, sorry, forgot to mention. This is for the review page. When we are
reviewing projects.

closed 2018-05-03 by danainschool