udacity/robotics-beta-test-issues #769: SLAM: L02E12 - Generate the map

zenetio, 2018-03-02:
The code is firing an error when it tries to save the map to disk.

https://classroom.udacity.com/nanodegrees/nd209-beta/parts/bece8892-4bc6-44e2-a5f6-53c2b3707ce4/modules/4b66b4d9-f728-4029-a7d1-9a77535dce8e/lessons/09ff9955-a893-43b2-a314-819a02f9231c/concepts/f8b5e95b-5d85-4020-9a3a-d52696a4a740

karimchamaa, 2018-03-02:
Can you show me the error? And are you using the workspace or the supplied VM?

zenetio, 2018-03-03:
I tried in both VM and workspace and the error is the same.
![image](https:
[image]
//user-images.githubusercontent.com/578337/36929715-aae8fb5e-1e73-11e8-99cd-

zenetio, 2018-03-03:
To check the code I added a show() before save and the picture was displayed correctly. But fails trying to save().

karimchamaa, 2018-03-03:
Thanks for sharing. Not sure why you are getting this or how I can replicate this error but I try in both VM and workspace and it both worked. I am afraid you are deleting the Images folder itself. 
Can you try these and let me know if you get it working?:
$ cd Downloads/
$ git clone https://github.com/udacity/RoboND-OccupancyGridMappingAlgorithm
$ cd RoboND-OccupancyGridMappingAlgorithm/
$ rm Images/*
$ g++ solution.cpp -o app -std=c++11 -I/usr/include/python2.7 -lpython2.7
$ ./app

zenetio, 2018-03-05:
Fixed the issue.

closed 2018-03-05 by zenetio

karimchamaa, 2018-03-05:
Still curious to know why you were getting the error?

zenetio, 2018-03-05:
There was a syntax error in the Image name string passed to save() function. To avoid this I should have tried the solution.cpp before open an issue. Tks.