udacity/robotics-beta-test-issues #779: HSR project - L08- Wall Follower

zenetio, 2018-04-24:
After run the wall follower I got a good map. But now when I try to save the file using the command rosrun map_server map_server mymap I get an error message saying that I am using a old map server interface and need switch to the new interface. What is this new interface that I could not find in internet?

karimchamaa, 2018-04-24:
Never seen that error, maybe run an update/upgrade?

zenetio, 2018-04-25:
Yes, I already did an upgrade with no success. This is weird because after got a good map definition, now I cannot save it.

zenetio, 2018-04-25:
Fixed!

karimchamaa, 2018-04-25:
How did you fix it ? And what do you think was the error?

zenetio, 2018-04-26:
The command line was wrong. I was ever typing map_server map_server when the correct command to save the map is map_server map_saver.

karimchamaa, 2018-04-26:
oh ok

closed 2018-04-26 by karimchamaa